FloatingArrayDesign/MoorDyn #271: incorrect Wave generation from wave frequency file

ewoudVL13, 2024-11-11:
Hello everybody, 
While doing validation from MoorDyn V2 results with OrcaFlex a basic setup of a line connected between two points and an irregular wave input.  
[image]
A big discrepancy in results occurred:
[image]
I can not find anything wrong in the input files and the wave frequency file is constructed as declared in the manual.
Does any of you have an idea whats going on? And also is there a way to get the surface height as an output from MoorDyn V2?
All the used input files:
[test_line.txt](https://github.com/user-attachments/files/17703831/test_line.txt)
[water_grid.txt](https://github.com/user-attachments/files/17703832/water_grid.txt)
[wave_frequencies.txt](https://github.com/user-attachments/files/17703833/wave_frequencies.txt)

RyanDavies19, 2024-11-11:
Hi @ewoudVL13,

I am not super familiar with WaveKin2 (@sanguinariojoe or @AlexWKinley might have better answers), but I can offer some troubleshooting ideas: You can try using [MoorDyn_GetWavesKin](https://moordyn.readthedocs.io/en/latest/api_c.html#_CPPv419MoorDyn_GetWavesKin12MoorDynWavesdddAL3E_dAL3E_dPdPd15MoorDynSeafloor), where the wave and seafloor instance can be found with [MoorDyn_GetWaves](https://moordyn.readthedocs.io/en/latest/api_c.html#_CPPv416MoorDyn_GetWaves7MoorDyn) and [MoorDyn_GetSeaFloor](https://moordyn.readthedocs.io/en/latest/api_c.html#_CPPv419MoorDyn_GetSeafloor7MoorDyn)

AlexWKinley, 2024-11-11:
A few quick thoughts.

I think you're wave file is slightly incorrect, in particular (and this is probably on me for writing unclear documentation), the first column of the wave_frequencies files is the angular frequency in radians per second.

https://moordyn.readthedocs.io/en/latest/waterkinematics.html#wavekin-2-wave-fft-grid
```lines containing a frequency (in radians per second) followed by the real and imaginary parts of the amplitude at that frequency```

So if the frequency is 0.061Hz, the angular frequency is $2\pi * 0.061 = 0.38327$. 

The other thing worth trying, is at least temporarily switching to wavekin = 7. This should do the exact sum of all the wave components at every node at every time step. As such, it does come with a performance hit, but it does handle irregular frequencies and should be able to give results more similar to orcaflex. Wavekin 2 internally interpolates your wave frequencies to be linearly spaced, so that it can use an fft-based calculation approach. 

Also worth noting that the way that OrcaFlex and MoorDyn do their wave calculations follow slightly different conventions, so you may not get the exact same wave timeseries out.
Python code that basically does the conversion between the wave data that orcaflex reports, into wave data for MoorDyn looks something like
```python
    freq_arr = waveSpectrum["Frequency"]
    amplitude_arr = waveSpectrum["Amplitude"]
    phase_arr = waveSpectrum["Phase"]
    direction_arr = waveSpectrum["Direction"]

    freq_arr = freq_arr * 2 * np.pi

    amp_arr = -amplitude_arr
    phase_arr = -np.deg2rad(phase_arr) - (np.pi / 2)

    re = amp_arr * np.cos(phase_arr)
    im = amp_arr * np.sin(phase_arr)

    rad_dir = np.deg2rad(direction_arr)

   # freq_arr = column 1
   # re = column 2
   # im = column 3
   # rad_dir =  optional column 4
```
between handling this convention difference, and using WaveKin = 7, you should be able to get the exact same wave timeseries between the two programs.

ewoudVL13, 2024-11-12:
[image]
@RyanDavies19 thx for the help, the results are now showing the same pattern. The angel orientation for rad_dir is as shown in the figure above. Just for info for others. pls check.

RyanDavies19, 2024-11-12:
Hi @ewoudVL13, 

That looks correct, but note that the units need to be radians rather than degrees. MoorDyn should throw an error to catch this. Thanks for documenting things here for future users!

RyanDavies19, 2024-11-25:
Hi @ewoudVL13, 

Are you running the python package? The only example I have on hand is in the tests, it is in C: https://github.com/FloatingArrayDesign/MoorDyn/blob/e5724689db20979e22db178b17519278d28be164/tests/c_api.c#L530

If you are in the python package, @sanguinariojoe might have more ideas.

ewoudVL13, 2024-11-25:
@RyanDavies19 yeah I 'm working in the python package. @sanguinariojoe do you know what's going on?

sanguinariojoe, 2024-11-25:
I'll check from wednesday on

sanguinariojoe, 2024-11-27:
Yep, there was a typo on `moordyn.GetWaveKin()` because  the time is not required anymore (as it is not required by `MoorDyn_GetWavesKin()`). It was indeed correct on the bindings, but I forgot to change that on the thin wrapper.

I really hope to find time to start adding Python tests like crazy. And MATLAB... And FORTRAN... And Rust!

RyanDavies19, 2024-11-27:
Thanks for the quick fix @sanguinariojoe!

ewoudVL13, 2024-11-27:
Wow so it works now?

RyanDavies19, 2024-11-27:
@ewoudVL13 try compiling the dev branch and building the python library, that will hopefully get things working for you.

sanguinariojoe, 2024-11-28:
It was not merged guys... Now you can try the dev branch.

We have been fixing a couple of things on the wrappers. Maybe we can launch a new minor release @RyanDavies19

sanguinariojoe, 2024-12-12:
Patch release is on its way!

closed 2025-04-22 by RyanDavies19