FloatingArrayDesign/MoorDyn #272: Discrepancy in results buoy in wave

ewoudVL13, 2024-11-14:
Hello everybody,
I am currently working on an model comparison between MoorDyn V2 and OrcaFlex. 
I ran a model of a line with 2 fixed end points in a regular wave and the results there are almost the same.  
[image]
Now I use the same wave input or a buoy with a center of mass that is lowered. (input files below)
[buoy_mass.txt](https://github.com/user-attachments/files/17750485/buoy_mass.txt)
[water_grid.txt](https://github.com/user-attachments/files/17750486/water_grid.txt)
[wave_frequencies.txt](https://github.com/user-attachments/files/17750487/wave_frequencies.txt)
But here something strange happens. The buoy is moving backwards. The usefful results are shown below. 
[image]
[image]
[image]
The buoy does not lay flat and for a basic calculation no cd or ca coefficients are included.
Any ideas?

RyanDavies19, 2024-11-25:
Hi @ewoudVL13, 

Sorry for the slow response here. It seems like your buoy has no hydrodynamic properties (Rod: Cd, Ca, CdEnd, and CaEnd). Is this intentional? My guess is this is causing the discrepancies.

ewoudVL13, 2024-11-25:
@RyanDavies19 Yeah it was done on purpose to see how the to programs act when only buoyancy is in the play.

ewoudVL13, 2024-11-25:
Yeah indeed I found that during the simulations. Dou you know something about the influence of wave height or wave period on the results?

RyanDavies19, 2024-11-25:
I'm not sure I can give specifics, but without the hydrodynamic forcing I would think the buoy would drift towards the anchor point rather than stay where it is. It could also be good to check decay tests of the buoy with no mooring, to make sure the dynamics are matching between the two (if results differ when including hydro stuff, then setting the water density to zero will determine if the kinematics are right). With no mooring and a cylindrical shape, pitch and heave should be sufficient.

RyanDavies19, 2025-03-03:
Closing due to inactivity

closed 2025-03-03 by RyanDavies19