FloatingArrayDesign/MoorDyn #273: Model of Single Point Mooring System

LaplaceSoda, 2024-11-17:
Hello everyone,

I am currently in the process of modelling a Single Point Mooring System, whose configuration can be written as
```
---------------------- POINT PROPERTIES --------------------------------
ID    Type      X       Y       Z       Mass   Volume  CdA    Ca
(#)   (-)       (m)     (m)     (m)     (kg)   (mˆ3)   (m^2)  (-)
1     Fixed     -400    0.0     -50.0    0      0       0      0
2     Fixed     -300    0.0     -50.0    0      0       0      0
3     Fixed     400     0.0     -50.0    0      0       0      0
4     Vessel    0.0     0.0      0.0     0      0       0      0
---------------------- LINES ----------------------------------------
ID   LineType   AttachA  AttachB  UnstrLen  NumSegs  LineOutputs
(#)   (name)     (#)      (#)       (m)       (-)     (-)
1     chain      1        4         410       82      pU
2     chain      2        4         410       82      pU
3     chain      3        4         410       82      pU
---------------------- OPTIONS -----------------------------------------
```
My questions are that

1. With the above configuration, the number of coupled DOFs seems to be 3. Are the output forces `f` in `MoorDyn_Step` function the sum of the forces of the three lines at the `#4` point?

2. If there is no special treatment for single point mooring system, the configuration file can also be written as follows
```
---------------------- POINT PROPERTIES --------------------------------
ID    Type      X       Y       Z       Mass   Volume  CdA    Ca
(#)   (-)       (m)     (m)     (m)     (kg)   (mˆ3)   (m^2)  (-)
1     Fixed     -400    0.0     -50.0    0      0       0      0
2     Fixed     -300    0.0     -50.0    0      0       0      0
3     Fixed     400     0.0     -50.0    0      0       0      0
4     Vessel    0.0     0.0      0.0     0      0       0      0
5     Vessel    0.0     0.0      0.0     0      0       0      0
6     Vessel    0.0     0.0      0.0     0      0       0      0
---------------------- LINES ----------------------------------------
ID   LineType   AttachA  AttachB  UnstrLen  NumSegs  LineOutputs
(#)   (name)     (#)      (#)       (m)       (-)     (-)
1     chain      1        4         410       82      pU
2     chain      2        5         410       82      pU
3     chain      3        6         410       82      pU
---------------------- OPTIONS -----------------------------------------
```
In this case, the the number of coupled DOFs increases to 9. How does this approach differ from the previous one? Additionally,  the output forces `f` in `MoorDyn_Step` function will also change to 9 dimensions,  do they represent the forces of each line at the place of `#4` point?

I appreciate any guidance or clarification on these points.

Thank you!

sanguinariojoe, 2024-11-17:
Hey!

Yes, on the first case you have a single coupled point (the 4th one), and
thus you have 3DOFs (x, y and z coords of that point). And yes,
MoorDyn_Step() is giving you back the net force on that point.

The difference on the second case is that the 3 points can be moved
independently from each other, and thus you have 9 DOFs (3 sets of x,y and
z coords). On that case MoorDyn_Step() is returning you the net force on
each of those points (9 components therefore).

Some hints:

- If you are interested on getting the forces on each line, you do not need
to use 3 coupled points. You can use the API entries, like
MoorDyn_GetLineNodeForce()
- There is very little to none difference on performance between both
approaches.
- If you have different fairleads for each line, but they are all attached
to the same rigid solid, you can use a body to move them. See the tests

sanguinariojoe, 2024-11-18:
> Thank you for your kind response.
> In fact, in order to figure out my problem, I wrote a coupling program in
> C++ and conducted some tests. In the following codes, the forces on the
> coupled point at the first time step was calculated.
> #include "third_party/MoorDyn/MoorDyn2.h"
> #include <iostream>
> void OutputForcesResults(const char *filename);
> int main()
>     std::cout << "---------------------------Case 1!---------------------------" << std::endl;
>     OutputForcesResults("..\\lines1.txt");
>     std::cout << "---------------------------Case 2!---------------------------" << std::endl;
>     OutputForcesResults("..\\lines2.txt");
>     return 0;
> void OutputForcesResults(const char *filename)
>     MoorDyn system = MoorDyn_Create(filename);
>     std::cout << "System created" << std::endl;
>     unsigned int n; // Number of coupled DOFs
>     int ret = MoorDyn_NCoupledDOF(system, &n);
>     if (ret != MOORDYN_SUCCESS)
>         std::cerr << "Error getting number of coupled DOFs" << std::endl;
>     double *x = new double[n]();
>     double *xd = new double[n]();
>     for (int i = 2; i < n; i += 3)
>         x[i] = -70.0;
>     std::cout << "x = ";
> [... 55 lines cut ...]
> -86.8127 0 -1.80779e+06
> 866388 0 -602583
> 0 -602583 -433237
> -602583 -433237 750381
> What is going on here? Did I make a mistake somewhere?

Regarding the initialization, try adding the following option on the config
files (lines*.txt):

1 ICgenDynamic Legacy initialization method

Regarding the output forces, the following line is wrong:

std::cout << f[i + j] << " ";

I think the indexing should read f[i * 3 + j]. I did not check that though

LaplaceSoda, 2024-11-18:
> Regarding the output forces, the following line is wrong: std::cout << f[i + j] << " "

Yes, you are right. Actually I do have noticed my stupid question and gotten the correct results. Thank you so much.
> The following result of the output forces `f` at coupled point using the two methods above has proved the correctness of your statement.
> ```
> t = 0.002 f =
> -86.8127 0 -1.80779e+06
> ......
> t = 0.002 f =
> 866388 0 -602583
> -433237 750381 -602605
> -433237 -750381 -602605
> ```

**Regarding the initialization**
> try adding the following option on the config files (lines*.txt): 1 ICgenDynamic Legacy initialization method

The system seems to converge faster using the old method, and the results don't appear to differ much from those of the new method.
```
Fairlead tensions converged
Remaining error after 0.1 s = 0.066998% on line 1
```
I wonder if the message `Remaining error after 76 s = XXX m/s2. Best score at 76 s = XXX m/s2` represents the non-convergence. I noticed in the documentation that 'The new stationary solver in MoorDyn-C is more stable and more precise than the dynamic solver, but it can take longer to reach equilibrium'. However, in my case, the convergence of the new method seems to worsen, and it also takes longer time.

In addition, could you help with some of the questions I asked earlier?

> 3\. In some coupling codes with Moordyn V1, I noticed that the dimensions of `x`, `xd`, and `f` don't seem to strictly equal to the `NCoupledDOF`. They might be directly set to 6 dimensions. Is that correct? How will it affect the results?

> in my understanding, the input `x` and `xd` are the position and velocity at the current time **_t0_**, and the output `f` repsent the forces at the next time step **_t0 + dt_** in function `MoorDyn_Step`， which means that when first calling `MoorDyn_Step`， the `x` should be equal to the `x` in `MoorDyn_Init`. Is that correct?

sanguinariojoe, 2024-11-18:
I noticed that the new method usually enjoys longer time steps, and thus
works better with implicit schemes.

Anyway, combining both methods seems to be the winning horse. You can find
a example on the docs.

Regarding question 3. First, forget about V1. It is outdated and
unsupported. That includes config files.

When you call MoorDyn_Step(), x is indeed the position at t0 (you really
must have a good reason to consider something else). xd is a bit more
complex. For instance, for prescribed motions xd = (x(t0 + dt) - x(t0)) /
dt.
And you are right about the returned forces, they are the ones at t0 + dt.

LaplaceSoda, 2024-11-21:
> I noticed that the new method usually enjoys longer time steps, and thus works better with implicit schemes. Anyway, combining both methods seems to be the winning horse. You can find a example on the docs.

Ok, thank you very much. I will check the documentation in more detail later.

> Regarding question 3. First, forget about V1. It is outdated and unsupported. That includes config files.

Regarding my third question, my finding is that although MoorDyn V1 does not have the `Bodies` class, it seems to implicitly include the concept of the bodies. In that case, as you mentioned at the beginning, it seems normal for both `x` and `xd` to be 6-dimensional. Of course, this is just my guess and cannot be confirmed.

> When you call MoorDyn_Step(), x is indeed the position at t0 (you really must have a good reason to consider something else). xd is a bit more complex. For instance, for prescribed motions xd = (x(t0 + dt) - x(t0)) / dt.

Thank you for your answer. I think I roughly understand your point. Now, a new question is, if it's not a prescribed motion (such as roll decay), what should be done then? If the time step _**dt**_ is small, could the `xd` at _**t0**_ be used instead?

sanguinariojoe, 2024-11-21:
> Now, a new question is, if it's not a prescribed motion (such as roll
decay), what should be done then? If the time step dt is small, could the
xd at t0 be used instead?

There is no a unique recipe. But yes, if dt is small xd at t0 will be for
sure a good approximation. However, keep in mind that x is still the
position at t0, AS IT IS INTEGRATED BY MOORDYN, i.e. x(t0 + dt) = x(t0) +
dt * xd(t0). That would not match the integrated position by your coupled
solver.

Thus, if you are willing to consider a staggered coupling, the usual
(consistent 1st order) way to go is:

1. You take the forces at t0 from MoorDyn.
2. You use them to integrate your motions solver
3. You take the positions at t0 + dt from your solver
4. You compute xd as (x(t0 + dt) - x(t0)) / dt
5. You integrate MoorDyn calling to MoorDyn_Step()

I personally recommend using MoorDyn_GetPointPos() to get x(t0), so you are
sure no inconsistencies/unstabilities are accidentally introduced there

LaplaceSoda, 2024-11-24:
> @LaplaceSoda, to your question on MoorDyn v1: V1 used to only couple via a 6 DOF body to external code, and there was no flexibility in the type and number of coupling points to external code. So your guess is right, and equivalent behavior can be simulated in MD-V2 with a zero mass, zero volume, and zero coefficient coupled body. However as @sanguinariojoe said, v1 is an old code that is not as accurate or reliable as MD-V2. We highly recommend users discontinue use of MD-V1.

@RyanDavies19 Thanks for your detailed clarification! I'll be using MD-V2 in the latest code.

LaplaceSoda, 2024-11-24:
> 1. You take the forces at t0 from MoorDyn. 2. You use them to integrate your motions solver 3. You take the positions at t0 + dt from your solver 4. You compute xd as (x(t0 + dt) - x(t0)) / dt 5. You integrate MoorDyn calling to MoorDyn_Step()

This sounds like a perfect solution. At this point, `xd` seems to represent the concept of average velocity considering acceleration. Thank you very much! :smile:

> I personally recommend using MoorDyn_GetPointPos() to get x(t0), so you are sure no inconsistencies/unstabilities are accidentally introduced there

If I ensure that x(0) is equal in all solvers when t = 0, then the `x(t0)` obtained from `MoorDyn_GetPointPos()` should strictly match the values from the coupled solver. So I wonder what could lead to inconsistencies/instabilities?

sanguinariojoe, 2024-11-24:
> If I ensure that x(0) is equal in all solvers when t = 0, then the x(t0)
obtained from MoorDyn_GetPointPos() should strictly match the values from
the coupled solver. So I wonder what could lead to
inconsistencies/instabilities?

I said accidentally. Let's say you do something wrong, like coupling a
solver with a patriot's missile bug. Quite subtle... Everything works
rather fine, but suddenly MoorDyn start giving back noisy forces... You
have no clue what is going on... Who is to blame? MoorDyn? Your solver? The
coupling? The MoorDyn inputs? The solver inputs? How to debug it?

If you followed my advice, then you know the problem is either coming from
your solver or from the staggered coupling (it might get unstable)

LaplaceSoda, 2024-11-25:
> I said accidentally. Let's say you do something wrong, like coupling a solver with a patriot's missile bug. Quite subtle... Everything works rather fine, but suddenly MoorDyn start giving back noisy [...]

OK .I think I roughly understand what you mean. Thank you!

Additionally, I would also like to know whether, at time t=0, after executing `MoorDyn_Init`, it is necessary to pass the mooring forces (obtained through `MoorDyn_GetBodyForce`) to the coupled solver. If the system is in equilibrium initially, then there is no need to pass the forces. However, if the system is inherently unbalanced at the start (e.g., due to heave decay), should I apply the forces obtained after executing `MoorDyn_Init` to the system?

LaplaceSoda, 2024-11-25:
I have some new ideas, but I'm not sure if they're correct. 
If the system is initially in equilibrium, the mooring forces are typically symmetrically distributed, and the resultant force in the horizontal plane is zero, with only vertical forces remaining. In this case, the vertical mooring force still needs to be considered and passed to the coupled solver.
So, even in the case of initial equilibrium, the mooring forces should still be passed to the coupled solver at the initial time step.

sanguinariojoe, 2024-11-27:
In general I recommend you to do not consider very particular cases if it is not strictly required. They will make your life way harder...

Just pass the forces at the beginning and be happy

LaplaceSoda, 2024-12-04:
> In general I recommend you to do not consider very particular cases if it is not strictly required. They will make your life way harder...
> 
> Just pass the forces at the beginning and be happy 

Okay, thank you very much, I'll follow your advice. 

Thank you again for your continued patient help and guidance.

sanguinariojoe, 2024-12-04:
> @sanguinariojoe <https://github.com/sanguinariojoe> Thank you again for

You are welcome!

closed 2024-12-05 by LaplaceSoda