FloatingArrayDesign/MoorDyn #274: Problems encountered when compiling V2 for matlab

frozenheartsaber, 2024-11-17:
Hello everybody

I am trying to use Moordyn with my code in matlab. I tried to compile it as mentioned in : https://moordyn.readthedocs.io/en/latest/compiling.html#id4  
I installed Cmake, MinGW and added them into path. I find Cmake ask me to install a fortran installer which I fixed by installing mingw-w64-x86_64-gcc-fortran in MINGW64. But then Cmake shows :" PyPA’s build package is required. Please install it typing msys64/mingw64/bin/python3.12.exe -m pip install --upgrade build". I use cmd to install it and Cmake seems to run well, but when I use:
mingw32-make, it fails as shown in picture. I wonder how to fix it? I tried to reinstall mingw or cmake, nothing works.
[image]

sanguinariojoe, 2024-11-18:
Just disable the Python wrapper on CMake setting the option PYTHON_WRAPPER
as OFF

frozenheartsaber, 2024-11-18:
> Just disable the Python wrapper on CMake setting the option PYTHON_WRAPPER as OFF
> […](#)

I installed it and found only libmoordyn.dll in the source. As shown on the website, there should be libmoordyn、libmoordyn.2 and libmoordyn.2.2.

I added the path C:\Program Files (x86)MoorDynbin into Matlab, and I copied the code given in Matlab, but it could not run, I doubt if I forgot something. I hope to fix this problem.

[image]
[image]
[image]

frozenheartsaber, 2024-11-18:
> 你好@frozenheartsaber，
> 
> 您添加的路径似乎不包含 MATLAB 包装器文件。请尝试专门添加`MoorDyn/build/wrappers/matlab`到 MATLAB 路径。该目录应该包含每个可调用的 MoorDyn 函数的文件。

Hello RyanDavies19.

Thank you for your advice. I have just find this, I never think about that before. But it seems something is still missing :
Invalid MEX-file 'D:\Moordyn\MoorDyn.build\wrappers\matlab\MoorDynM_Create.mexw64': The specified module could not be found.
An error comes like this, I have changed the path as shown in the picture.
[image]

frozenheartsaber, 2024-11-18:
> 你好@frozenheartsaber，
> 
> 您添加的路径似乎不包含 MATLAB 包装器文件。请尝试专门添加`MoorDyn/build/wrappers/matlab`到 MATLAB 路径。该目录应该包含每个可调用的 MoorDyn 函数的文件。

Hi RyanDavies
I find those mex files in `MoorDyn/build/wrappers/matlab` are also included in C:\Program Files (x86)\Moordyn\bin. I added it into path, something went wrong without specific advice. It just tells me "Incorrect using". I hope to know what I shoud do to fix it.
[image]
[image]

RyanDavies19, 2024-11-18:
Hi @frozenheartsaber, 

Can you share your MATLAB code and MoorDyn input file? I suspect things are linked up correctly but MoorDyn is being used incorrectly

frozenheartsaber, 2024-11-18:
> Hi @frozenheartsaber,
> 
> Can you share your MATLAB code and MoorDyn input file? I suspect things are linked up correctly but MoorDyn is being used incorrectly

I find the reason, so sorry for my last question. I forgot I need to create a line.txt in this example. I will try to change my old .txt here to test again. Really thank you for your advice.

frozenheartsaber, 2024-11-19:
> 你好@frozenheartsaber，
> 
> 您能分享一下您的 MATLAB 代码（如果与上面分享的代码不同）和 MoorDyn 输入文件吗？我怀疑一切都正确连接了，但 MoorDyn 的使用方式不正确

Hi,RyanDavies
I tried to change a old txt and I hope to compare the result with Orcaflex. I find some problems during the test.  I modelled a 1000 seconds tests with and without current. I did a similar test with codes shown in figure below in the past as I need to couple the motion of the fairlead with a rigid body, I changes x and xd each time step.
[image]
And I hope to know if my new code means the same? I think it works well, but I am not so sure.
[image]
One of the problem is that, in the new code , I use f in  [t, f(i,1:3)] = MoorDynM_Step(system, x, xd, t, dt) and ten = MoorDynM_GetLineNodeTen(line, node_id-1) to get the tension in the fairlead but the results seems to be different. As I seen in the document , the tension at fairlead should be the same as I get from MoorDynM_Step. The results is shown in figure below
[image] The difference becomes more obviously when the the number of anchor chain sections is smaller。
Another problem is I can not use " 1  Currents the water currents are provided in a grid (-)" or using implicit method as wilison in current code which may cause an error. I wonder how to fix those problems.

[testmoor.zip](https://github.com/user-attachments/files/17807484/testmoor.zip)

frozenheartsaber, 2024-11-20:
Another problem is that, during the test above, I found the result different with Orcaflex when using catenary, the tension at the fairlead seems to be fun. But  tensions of lines from the anchor point to the touch down point are different in Moordyn and Orcaflex. I think it might be caused by the friction force? As I changed the shear stiffness of the seabed  to 0, the result seemed to be similar in Orcaflex.  I changed the cbot and friction coefficient in .txt, but nothing changed.

sanguinariojoe, 2024-11-20:
Yeah, MoorDyn and Orcaflex have different seabed friction models. Just look
for some literature about that on Google Scholar.

I tend to think that the Orcaflex model is rather inconsistent. But to be
honest, I do not know any paper with an actual validation. Not for Orcaflex
nor for MoorDyn. So take my comment just as an opinion.

sanguinariojoe, 2024-11-20:
P.S. Another difference that shall be incidentally considered, which might
significantly affect the results on some dynamic cases, is the time
integration scheme. OrcaFlex considers a quite dissipative (rho=0.4)
generalized-alpha scheme. I sometimes needed to increase the internal
damping to compensate that.

RyanDavies19, 2024-11-20:
> Yeah, MoorDyn and Orcaflex have different seabed friction models. Just look for some literature about that on Google Scholar. I tend to think that the Orcaflex model is rather inconsistent. But to [...]

I'll point to the references in the documentation for some papers regarding the seabed contact approaches used in MoorDyn: https://moordyn.readthedocs.io/en/latest/references.html#references

However note that the axial and transverse friction as described in https://doi.org/10.1088/1742-6596/2362/1/012018 have only been implemented in MD-F.

frozenheartsaber, 2024-11-20:
Thank you for your advice.  I think the main problem will be I can not use current or wave by changing 0 or 1 in .txt, so as changing to the implicit time integration method. And I  use my old code to make an example to do another test with current as given in the following zips. (results saves the .mat)

[1m_s_test_orca_compare.zip](https://github.com/user-attachments/files/17834470/1m_s_test_orca_compare.zip)
[results.zip](https://github.com/user-attachments/files/17834481/results.zip)

It can be found with figures below, at the begining of the simulation, an unexpected large force occurs in Moordyn, I think it is because Orcaflex did a static simulation before the dynamic simulation. I hope to try it in Moordyn. But as I can not use the new code with current, I have not tried it. And I think as said in the document, Moordyn has already used catenary solver for static caculation? In order to avoid numerical damping , I changed orcaflex into explicit calculation.

But It can also be seen that, the anchor force seems to be pretty different as I mentioned  before, as I know it may be caused by different firiction from your suggestion. But no matter I change which parameter, the anchor force seems do not change. I wonder how to solve above problems, I do not know if it is because I am using matlab version or I missed something else?
[image]
[image]

RyanDavies19, 2024-11-25:
Hi @frozenheartsaber, 

I'm not sure I understand your question about currents and time integrators. 

A couple things to check: Does MoorDyn reach equilibrium before the simulation states? If it fails to initialized you will get a warning in the console like: "Fairlead tensions did not converge". The initial force you see is likely due to a failure to find the steady state. Secondly, it seems that after this initial transient, the anchor force (Force on Line end A) converges on a similar value to your OrcaFlex outputs, just over 5e4 N. It might be helpful to output the motions of the fairlead point from MoorDyn using Point2PX/Y/Z. This will tell you if it is actually moving as you desire. Based on your outputs there, it seems the point is stationary.

RyanDavies19, 2024-11-25:
If you could provide the console outputs moordyn produces, or your log file, it would help figure out what's going on here.

frozenheartsaber, 2024-11-25:
> to check: Does MoorDyn reach equilibrium before the simulation states? If it fa
Hi RyanDavies,
The problem with current and time integrators is that I can not use implicit integrators or add current or wave conditions to mooring lines. I can only simulate mooring lines without current by using the new Matlab code, although I changed the options in .txt.
I put the log file and all inputs of this problem in the following problems_log_current_implicit.zip.
[problems_log_current_implicit.zip](https://github.com/user-attachments/files/17910187/problems_log_current_implicit.zip)

RyanDavies19, 2024-11-26:
Hi @frozenheartsaber, I tried replicating your error and got: 
```
ERR <path>/MoorDyn/source/Waves/WaveGrid.cpp:558 constructSteadyCurrentGrid(): Cannot read the file '/current_profile.txt'
Error (-1) during the Mooring System creation:
Exception while reading the input file
Error using testmoor (line 1)
MoorDyn reported an error
```

This might be a bug with the paths being picked up. I will take a look at it tomorrow. 

For dev notes: this is caused by the path passed to waves->setup: https://github.com/FloatingArrayDesign/MoorDyn/blob/2dadb165a1924aecb397ba719e8eabef3e52f792/source/MoorDyn2.cpp#L581

What issues have you had with the time integrators?

frozenheartsaber, 2024-11-26:
> s->s
I put the .log  with time integrator here. It shows "WRN D:\Moordyn\MoorDyn\source\MoorDyn2.cpp:2187 readOptionsLine(): Warning: Unrecognized option 'tscheme'"
[line_implicit.log](https://github.com/user-attachments/files/17917090/line_implicit.log)

sanguinariojoe, 2024-11-26:
Take care, It is case sensitive. Probably we should change that

frozenheartsaber, 2024-11-26:
> 注意，它区分大小写。也许我们应该改变这一点
> […](#)

Thank you for your advice. And I tried to change RK4 into rk4 or reverse， it works well. But no matter using  ACA5 or aca5 (given in the document), it can not find the right integrator.

RyanDavies19, 2024-11-26:
> Take care, It is case sensitive. Probably we should change that

Agreed, while I'm changing the options list in Alex's PR I'll add this in

RyanDavies19, 2024-11-26:
> > 注意，它区分大小写。也许我们应该改变这一点
> > […](#)
> 
> Thank you for your advice. And I tried to change RK4 into rk4 or reverse， it works well. But no matter using ACA5 or aca5 (given in the document), it can not find the right integrator.

Does moordyn say `Invalid Average Constant Acceleration name format` or `Unknown time scheme`?

RyanDavies19, 2024-11-26:
> > 你好@frozenheartsaber，
> > 我不确定我是否理解您有关当前和时间积分器的问题。
> > 需要检查以下几点：MoorDyn 是否处于模拟状态之前达到平衡？如果初始化失败，您将在控制台中收到警告，例如：“导缆器张力未收敛”。您看到的初始力可能是由于未能成功找到稳定状态。其次，似乎此时初始瞬态之后，锚力（线端A上的力）收敛到与OrcaFlex输出相似的值，略为5e4 N。使用来自MoorDyn的Point2PX/Y/Z输出导缆器点的运动可能会有所帮助。这将告诉您它是否真的按照您的意愿移动。根据您那里的输出，该点似乎是原来的。
> 
> I added the section and get a better results, but the initial problem is still here. I added the log file and source code below. I did not find the Point2PX/Y/Z so I used x(:,i) obtained from belowing code as an instead to show the position of the top node. "[~,~,x_moor(:,i),dx_moor(:,i),TopTension(:,i),time,delta_t] = calllib('libmoordyn2','MoorDyn_Step',system, x_moor(:,i), dx_moor(:,i), TopTension(:,i),time,delta_t);"
> 
> [matlab_code_old.zip](https://github.com/user-attachments/files/17917960/matlab_code_old.zip) [log_txt_profile_out.zip](https://github.com/user-attachments/files/17917961/log_txt_profile_out.zip)

@frozenheartsaber that approach will work for getting the node positions in real time. In your MoorDyn outputs you need to specify specifically POINT2PX, POINT2PY, or POINT2PZ. Just double checking are you saying that one of those output channels was not working?

frozenheartsaber, 2024-11-26:
> etter results, but the initial problem is still here. I added the log file and source code below. I did not find the Point2PX/Y/Z so I used x(:,i) obtained from belowing code as an instead to show [...]

Sorry，I misunderstand POINT2PX that I thought I should find it in functions, I will use it to get some output.

RyanDavies19, 2024-11-26:
Try tScheme

RyanDavies19, 2024-11-26:
> Hi @frozenheartsaber, I tried replicating your error and got:
>
> ```
> ERR <path>/MoorDyn/source/Waves/WaveGrid.cpp:558 constructSteadyCurrentGrid(): Cannot read the file '/current_profile.txt'
> Error [...]

PR #276 contains a fix to the path error you were seeing @frozenheartsaber. It was only appearing in the MATLAB wrapper, which we don't have a robust set of tests for because MATLAB is annoying

frozenheartsaber, 2024-12-01:
Thank you for your help! I will test it in next week. I was busy with some coupling problems last week.

sanguinariojoe, 2024-12-12:
@frozenheartsaber , can we close this?

frozenheartsaber, 2024-12-17:
So sorry for my late reply. I was writing a code of rigid body in the past weeks. . I will focus on running the new code .

closed 2024-12-17 by frozenheartsaber