FloatingArrayDesign/MoorDyn #278: Divergent branches

sanguinariojoe, 2024-11-28:
Hey @RyanDavies19 and @AlexWKinley !

For some reason (probably we rebased on master instead of merging) the dev branch has diverged from the master one. I rebased and the problem was automagically gone, so it is not really a big deal. However, if I do that I will break all your on-going work (including the active PRs).

Thus, we can proceed in two ways (I really think the first way is the good one):

1. Launch a new release forcily pushing the current dev on master
2. Rebase dev on master

The merit of way 1 is that everything will work just fine for all us. The problem is that we are forcily pushing on master, so some remote forks  on the wild might have problems to create PRs. I think the risk is rather low because we are actually the only active devs of this project. If someone complains on the close future I will take full responsibility on fixing the issue.

RyanDavies19, 2024-12-03:
I'm fine with force pushing to master. I am not aware of anyone doing and development, and if they are I would think they would be working off the dev branch.

sanguinariojoe, 2024-12-12:
OK, I force pushed. Patch release is on its way!

closed 2024-12-12 by sanguinariojoe