FloatingArrayDesign/MoorDyn #281: Compiling MoorDyn on Linux Issues

jtgrasb, 2024-12-09:
I am working on setting up MoorDyn to couple with WEC-Sim on different platforms (see [WEC-Sim MoorDyn repo](https://github.com/WEC-Sim/MoorDyn)). To generate a .so file (and test it), I am using Sandia's HPC servers, which I have to ssh into and have limited rights on. 
1. I tried using the [pre-compiled releases](https://github.com/FloatingArrayDesign/MoorDyn/releases/tag/v2.3.7), which worked to compile it and generate the .so file, but then I got a libvtk issue that prevented it from running. Since, you can't toggle the vtk on/off for the pre-compiled release, I moved on to trying the compile using cmake.
2. Using cmake, I ran into a few issues. First, I had to add #include <iomanip> to MoorDyn2.hpp so that `setprecision()` could recognized. Then, I also ran into a similar issue in one of the testing scripts so had to disable testing. The following command worked for compiling:
`cmake -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_BUILD_TYPE=Release -DPYTHON_WRAPPER=OFF -DBUILD_TESTING=OFF -DCMAKE_INSTALL_PREFIX=/gpfs/jtgrasb/MoorDyn_install/ ../`
But, after successfully compiling, I got an error while testing that said: "ELF load command past end of file" (brief research suggests corrupted library). 
3. I also tried running `make` from the SO folder, which had some issues with the `-static-libstdc++`  command. After removing that line from the script, I was able to get it to compile, but got the same "ELF load command past end of file" error when testing. 

@sanguinariojoe Do you all have any ideas what may be causing the issues? Originally, I thought it may be a compiler issue, but I am not entirely sure how to check which compiler is being used (because there are multiple installed). Let me know if you have any suggestions. 

--Thanks!
Jeff

RyanDavies19, 2024-12-09:
@sanguinariojoe for some more context, we are not using the MATLAB wrapper yet for WECSim because we want to ensure compatibility with a range of MATLAB versions while providing precompiled MD libraries. To my understanding the MATLAB wrapper is built for the version of MATLAB you have when compiling. We are working on automating this so WECSim just pulls the releases to the main branch here, but in the meantime we need a linux distribution while we work on that. This seemed like a compiler issue to me but I am not super familiar with Linux so I was hoping you might have some ideas.

sanguinariojoe, 2024-12-09:
Hey guys! I am coming back to the office soon. I will help you then.

Anyway, you can try to download the vtk libs here:

https://github.com/sanguinariojoe/vtk-builds/releases/tag/VTK-9.3.1-shared

And use them to try the precompiled libs again

jtgrasb, 2024-12-09:
@sanguinariojoe Thanks for the suggestion. I'm not too familiar with installing things on Linux. I unzipped the contents using `tar -zxvf vtk-manylinux_x86_64.tar.gz`, then do I need to install them using sudo? Unfortunately, I don't have admin privileges so can't use sudo and was hoping to be able to get MoorDyn compiled and working without needing sudo.

sanguinariojoe, 2024-12-09:
No need for sudo, just place the libs anywhere and do:

export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/the/folder/with/the/libs

Before executing whatever stuff that is using libmoordyn

jtgrasb, 2024-12-10:
@RyanDavies19 @sanguinariojoe Thank you for the suggestions. I actually realized that I was making a mistake on my end when testing the .so files. cmake creates 3 files: libmoordyn.so, libmoordyn.so.2, and libmoordyn.so.2.3. I was using libmoordyn.so while the correct file is libmoordyn.so.2.3. I have fixed this and updating the WEC-Sim/MoorDyn repository now using the cmake version (I don't want users to have to install vtk since its not being used for our case). Thanks!

RyanDavies19, 2024-12-10:
@jtgrasb, in theory all three of those are the same, as .so.2 and .so are alias's that point to .so.2.3. 

However hopefully we can get this working soon. @jtgrasb does WECSim have VTK support? For a later conversation, but some form of coupling there could be interesting too.

jtgrasb, 2024-12-10:
Good to know. I had also tried putting all 3 in the folder, but it only works when I change the name of libmoordyn.so.2.3 to libmoordyn.so (because that's the file WEC-Sim is looking for). But works well with just the one file.

No, WEC-Sim does support Paraview for visualization purposes (and can visualize the mooring lines in there), but not VTK. I can add that to our project board as its probably something that people would be interested in.

RyanDavies19, 2024-12-10:
@jtgrasb, only if there's time and availability to support it. I think the more important thing is automation of this build process if we can.

jtgrasb, 2024-12-10:
Agreed, I plan to start working on that next week.

sanguinariojoe, 2024-12-10:
Ryan is right, libmoordyn.so is a symbolic link of libmoordyn.so.2, which
is a symbolic link of libmoordyn.so.2.3.

Said that, sometimes to open/load a file you must specifically indicate to
follow symbolic links. Are you dynamically loading the library? Are you
linking wecsim with libmoordyn.so either?

sanguinariojoe, 2024-12-12:
I clsoe this. We can reopen if needed or follow up on a different issue

closed 2024-12-12 by sanguinariojoe

jtgrasb, 2024-12-12:
Sorry for the late reply, was caught up with some other tasks. Fine to close this as there are no longer any issues. But, to answer your questions, We are loading the library into MATLAB using the [loadlibrary()](https://www.mathworks.com/help/matlab/ref/loadlibrary.html) function. Then, We're using [calllib()](https://www.mathworks.com/help/matlab/ref/calllib.html) to initialize and run MoorDyn.

In terms of the symbolic link, what is the benefit of linking one file to another when libmoordyn.so.2.3 seems to work fine on its own?

sanguinariojoe, 2024-12-12:
Yep, MATLAB guys recommend using mex files instead of loadlibrary(),
because a lot of stuff is not supported. Anyway, I am quite sure it can
follow symbolic links. Have you tried not moving the files, but calling a
'cd' to go to the libs folder before calling loadlibrary()?

And about the symbolic links, good question indeed. It is a naming
convention that makes everything works as expected on Linux (which is a
gigantic task by the way). Let me try to elaborate the idea. We have 3
files:

libmoordyn.so
 L_ libmoordyn.so.2
       L_ libmoordyn.so.2.3

The 2 and 3 suffixes are the major and minor versions of the library. The
first file and the headers are usually called the development files,
whereas the last file is usually called the library.

When a developer needs moordyn for its project, he instructs the compiler
to link against libmoordyn.so, which will automatically follow up the
symbolic links to reach the file libmoordyn.so.2.3 . Indeed, if you query
the produced binary with ld, it will list libmoordyn.so.2.3 as a
dependency, not libmoordyn.so

Now let's say we discover an issue on moordyn, which usually does not
require adding/editing the API entries (the MoorDyn_* methods), i.e. it is
fixed editing the cpp files. Then we can correct the bug, increase the
patch version (so major and minor remain untouched), recompile and launch
the release. The developers DO NOT NEED to recompile their softwares with
the new library. People just need to install the new version of
libmoordyn.so.2.3 file and everything will continue working just fine.

Now let's say we discover something really wrong, or we just want to add a
new feature, that requires changing the API. Now we cannot proceed the same
way, because the binaries already linked against libmoordyn.so.2.3 will
crash. Thus we increase the minor version, producing slightly different
files:

libmoordyn.so
 L_ libmoordyn.so.2
       L_ libmoordyn.so.2.4

If someone install those new files, the former library libmoordyn.so.2.3
will not be overwritten, and thus all the binaries that where using moordyn
will happily continue working after the update. However, any developer can
install the new version, recompile his binary and then the compiler will
now link against libmoordyn.so.2.4 . Now he can distribute his new binary
which is using the updated moordyn. Other binaries will still work with
libmoordyn.so.2.3

So using this convention we can all be sure that updating a library will
not unexpectedly break programs, or even the system.

jtgrasb, 2024-12-12:
@sanguinariojoe Thank you for the detailed explanation! First, we had tried the mex file approach but felt that the loadlibrary approach was simpler from the user perspective who doesn't need to access the individual functions of MoorDyn and just needs WEC-SIm to call the library under the hood. 

As for loading the library, I did try to cd into the library's directory and call `loadlibrary('libmoordyn.so','MoorDyn.h');` with all 3 files present, but I still get the same "ELF load command past end of file" error.