FloatingArrayDesign/MoorDyn #282: Running MoorDynV2 in matlab

s-zhang20, 2024-12-20:
Hello everybody,

I am trying to use MoordynV2 within matlab. I have compiled it as mentioned in : https://moordyn.readthedocs.io/en/latest/compiling.html#id4

But when I ran the test case in matlab, it told "_Invalid MEX-file 'C:\MoorDyn\MoorDyn.build\wrappers\matlab\MoorDynM_Create.mexw64': The specified
module could not be found.

Error in main (line 5)
system = MoorDynM_Create('Mooring/lines.txt');_".

[image]

I hope to know what I shoud do to fix it.
I would be grateful for any hints.

Best wishes

RyanDavies19, 2024-12-20:
Hi @s-zhang20,

This error seems like your mex files were never made. Have you enabled the Matlab wrapper with CMake? A quick check would be to look at the content of your `C:\MoorDyn\MoorDyn.build\wrappers\matlab\` folder. There should be a number of `.mexw64` files in there, one for each function you can call.

s-zhang20, 2024-12-21:
@RyanDavies19 Thank you very much for your quick response, I appreciate it. I think I have enabled the Matlab wrapper with CMake, as shown in the following figure.

[image]

But when I call these functions in MATLAB, it says "Invalid MEX-file ...".

RyanDavies19, 2024-12-24:
Hi @s-zhang20 

This seems like a path error. If you are sure you have added the folder with the mexw64 files to the matlab path and you still get the error. I would suggest deleting your build directory (i.e. deleting the mexw64 files), and recompiling to build the mexw64 files again.

RyanDavies19, 2025-01-13:
Hi @s-zhang20, were you able to resolve this?

RyanDavies19, 2025-03-03:
Closing due to inactivity

closed 2025-03-03 by RyanDavies19