FloatingArrayDesign/MoorDyn #285: Installing pre-compiled releases using Github Actions

jtgrasb, 2025-01-08:
Hi all, I am trying to install the pre-compiled releases in GitHub Actions in order to keep the [WEC-Sim/MoorDyn](https://github.com/WEC-Sim/MoorDyn) repository updated with this one. I am still getting familiar with Github Actions, but I am testing the options on some private repositories. Here is my example code so far:

`name: Compile MoorDyn Libraries
on:
  workflow_dispatch:
  repository_dispatch:
    types:
      - run_compile
jobs:
  pull-file:
    runs-on: ubuntu-latest
    steps:
      - name: Download release asset
        uses: dsaltares/fetch-gh-release-asset@master
        with:
          repo: 'jtgrasb/test'
          version: 'tags/1'
          file: 'Moordyn-2.3.7-win64.exe'
          token: ${{ secrets.REPO_ACCESS_TOKEN }}

      - name: Upload file for next job
        uses: actions/upload-artifact@v4
        with:
          name: asset_exe
          path: Moordyn-2.3.7-win64.exe
        
  run-exe:
    needs: pull-file
    runs-on: windows-latest
    steps:
      - name: Download file from previous job
        uses: actions/download-artifact@v4
        with:
          name: asset_exe
      - name: List workspace
        run: |
          dir 
      - name: Create output directory
        run: |
          mkdir output
      - name: Run exe
        run: |
          .\Moordyn-2.3.7-win64.exe --output-dir=.\output
          echo "Exit code: $?"
      - name: List workspace
        run: |
          dir .\output
`

The exe seems to be running, but the output directory is not being populated. Is there another command I need to include to access the outputs of the exe? @RyanDavies19 Let me know if you have any suggestions.

RyanDavies19, 2025-01-08:
@jtgrasb great to see this is being worked on! I don't see anything off the top of my head that would cause issues. Does the `.\Moordyn-2.3.7-win64.exe --output-dir=.\output` work when you run it locally on your windows? There might me some more inputs the exe needs (i.e. when installing on Mac after executing it asks what folder to send things to and to accept a license).

jtgrasb, 2025-01-08:
No, it seems to output the installation to Program Files by default unless you change it using the windows installation GUI. I'm not sure how to see customization options for the exe either.

RyanDavies19, 2025-01-08:
@sanguinariojoe, would you have ideas on why the outputs aren't appearing? I know you set up the building of these distributions. 

@jtgrasb I'm not sure if have an answer then, other than more trial and error. At least that's what I would do. On a related note, did you have some form of testing you were thinking of for this?

sanguinariojoe, 2025-01-09:
Hey guys! Sorry, I caught a virus which is keeping me offline

I meant the windows installer for users, not CI/CDs... I suppose we can
also provide a zip file with libs and headers.

Anyway, you can try the option /silent when executing the installer. It is
installing it on the default folder (C:\\Program...), so you'll need to
move the files afterwards.

Never tried it though...

jtgrasb, 2025-01-09:
Thanks for the responses. I was able to do some more testing and could access Program Files to download and add the dll to my repository successfully! I will work on Mac and Linux as well to get them working on my test repo. My planned approach is:

1. Place a trigger.yml file on FloatingArrayDesign/MoorDyn which triggers an event on WEC-Sim/MoorDyn anytime a release is published. 
2. Place a compile.yml on WEC-Sim/MoorDyn which extracts the release assets, executes each one of them, then adds the compiled libraries to the repository.
    - This will require a secret access token to extract the assets from FloatingArrayDesign/MoorDyn. Can you generate this?
4. Add a trigger.yml that triggers the MoorDyn test to run on WEC-Sim/WEC-Sim_Applications.

Let me know if you have any suggestions for this approach.
Hope you feel better soon @sanguinariojoe

RyanDavies19, 2025-01-09:
@jtgrasb, 

That sounds like a good plan to me. I can generate the token as needed, what timeline are you looking at for that? We can discuss on teams and I can share things there

sanguinariojoe, 2025-01-10:
Hey guys!

I find all this quite convoluted and dangerous.

First, I do not understand why do you want to do all this every single time
we publish a release. In general, what you want to do is anchoring your
development (or the CI/CD) to a very specific version of the dependency
(MoorDyn in this case). Otherwise you will have big issues if we release
changes that break the API (so we are all probably getting flooded by CI/CD
errors spam).

Second, it is not way more sensible to provide users/consumers with
whatever they need (including needed stuff for CI/CDs) rather than "giving
away secrets"? I really trust on you, but this sounds like quite bad
practices and kind of one-time solution. Are we doing this each time
someone wants to integrate us on a CI/CD?

I think the best we can do is either:

- Add a ZIP with the headers and the libraries to the releases
- Create an action that users can import on their pipelines.

I do not know... Is it because of the fever?

RyanDavies19, 2025-01-10:
@sanguinariojoe thanks for the feedback on this. @jtgrasb is leading this effort, so I will let him drive what needs to be done for the automation from the WECSim side. For more context, it is a big priority for NREL and SNL to keep the WECSim MoorDyn-C coupling up-to-date because that is the biggest dependency of MoorDyn-C on our side (and thus the biggest supporter of MoorDyn-C work).

sanguinariojoe, 2025-01-10:
> from the WECSim side. It is a big priority for NREL and SNL to keep the

Well, your way is indeed making my life easier. If it turns to be
problematic it can be always fixed on the future... The bright side of the
computer sciences, right?

I just feel things can be done on a simpler and nicer way. I added my two
cents anyway, and I will be OK with whatever you decide

closed 2025-03-26 by jtgrasb