FloatingArrayDesign/MoorDyn #292: Inquiry About Configuring Gravity Direction in MoorDyn v2 for Coupled Simulations

wulizhiren, 2025-03-02:
Dear MoorDyn Developers and Community,

I am currently working on integrating MoorDyn v2 with a structural solver for coupled simulations. While reviewing the documentation and testing the system, I noticed that MoorDyn v2 defines gravity along the z-axis by default. However, my structural solver assumes gravity acts along the y-axis, which unfortunately creates inconsistencies in our coupled analysis.

Could you kindly advise whether MoorDyn v2 supports customizing the gravity direction (e.g., changing it from z-axis to y-axis)? If this functionality exists, I would greatly appreciate any instructions on how to configure it properly.

RyanDavies19, 2025-03-03:
Hi @wulizhiren, 

MoorDyn does not allow for custom reference frames, and gravity is hard coded to the z direction. For your coupled structural solver, you could add a step that transforms from your solver frame to the MoorDyn frame using a rotation matrix with a small amount of effort. 

If you are looking for inspiration, we do this type of reference frame transformation for 3 DOF points from the body reference frame to the global frame using the transform kinematics function: 
https://github.com/FloatingArrayDesign/MoorDyn/blob/4828499f8ade98521011f563c99fbd71c0429647/source/Misc.hpp#L889
https://github.com/FloatingArrayDesign/MoorDyn/blob/4828499f8ade98521011f563c99fbd71c0429647/source/Misc.cpp#L289

wulizhiren, 2025-03-05:
Dear @RyanDavies19 ,

Thank you so much for your prompt and detailed guidance! Following your suggestions, I successfully resolved the gravity direction alignment issue.The coupled simulations now run smoothly with consistent gravity definitions across both solvers. Your expertise saved me significant time and effort, and I truly appreciate your support.

closed 2025-03-05 by wulizhiren