FloatingArrayDesign/MoorDyn #298: Seeking Help with MoorDyn-Structural Solver Coupling: State Restoration Error After ~2000 Steps

wulizhiren, 2025-03-22:
Dear MoorDyn Developers/Community,

I'm working on coupling MoorDyn (v2.x) with a custom structural solver using the Python interface. While initial integration was successful via pip install moordyn (following [documentation](https://moordyn.readthedocs.io/en/master/compiling.html#install-as-python-module) ), I've encountered a critical issue during state restoration cycles and would appreciate your expertise.

Problem Description
After approximately 2000 simulation steps involving repeated SaveState/LoadState operations, the Python kernel fails with:
```log
moordyn: load old state
Fatal Python error: none_dealloc: deallocating None
Python runtime state: initialized

Current thread 0x00007d8dbb1c4740 (most recent call first):
  File "/usr/local/lib/python3.10/dist-packages/moordyn/moordyn.py", line 638 in LoadState
  File "Mypythonfile", line 150  #This line of code is md.LoadState(self.system,"oldstate")

Extension modules: numpy.core._multiarray_umath, numpy.core._multiarray_tests, numpy.linalg.lapack_lite, numpy.linalg._umath_linalg, numpy.fft._pocketfft_internal, numpy.random._common, numpy.random.bit_generator, numpy.random._bounded_integers, numpy.random._mt19937, numpy.random.mtrand, numpy.random._philox, numpy.random._pcg64, numpy.random._sfc64, numpy.random._generator, mpi4py.MPI, yaml._yaml, cmoordyn (total: 17)
*** Process received signal ***

Signal: Aborted (6)
Signal code:  (-6)
[ 0] /lib/x86_64-linux-gnu/libc.so.6(+0x42520)[0x7d8dbae42520]
[ 1] /lib/x86_64-linux-gnu/libc.so.6(pthread_kill+0x12c)[0x7d8dbae969fc]
[ 2] /lib/x86_64-linux-gnu/libc.so.6(raise+0x16)[0x7d8dbae42476]
[ 3] /lib/x86_64-linux-gnu/libc.so.6(abort+0xd3)[0x7d8dbae287f3]
[ 4] python3(+0x71e47)[0x5ddbb9978e47]
[ 5] python3(_Py_FatalError_TstateNULL+0x0)[0x5ddbb9a49825]
[ 6] python3(+0x14021c)[0x5ddbb9a4721c]
[ 7] python3(+0xa8634)[0x5ddbb99af634]
[ 8] python3(+0x1e9c9c)[0x5ddbb9af0c9c]
[ 9] python3(+0x18ec28)[0x5ddbb9a95c28]
[10] python3(+0x170b28)[0x5ddbb9a77b28]
[11] python3(+0x20bcbb)[0x5ddbb9b12cbb]
[12] python3(+0x18ec28)[0x5ddbb9a95c28]
[13] python3(+0x170b28)[0x5ddbb9a77b28]
[14] python3(+0x270e67)[0x5ddbb9b77e67]
[15] python3(+0x72183)[0x5ddbb9979183]
[16] python3(_Py_FatalError_TstateNULL+0x0)[0x5ddbb9a49825]
[17] python3(+0x14021c)[0x5ddbb9a4721c]
[18] python3(_PyEval_EvalFrameDefault+0x6029)[0x5ddbb9a82299]
[19] python3(_PyFunction_Vectorcall+0x7c)[0x5ddbb9a9266c]
[20] python3(_PyEval_EvalFrameDefault+0x5642)[0x5ddbb9a818b2]
[21] python3(_PyFunction_Vectorcall+0x7c)[0x5ddbb9a9266c]
[22] python3(_PyEval_EvalFrameDefault+0x804)[0x5ddbb9a7ca74]
[23] python3(+0x259f56)[0x5ddbb9b60f56]
[24] python3(PyEval_EvalCode+0x86)[0x5ddbb9b60e26]
[25] python3(+0x280808)[0x5ddbb9b87808]
[26] python3(+0x27b00f)[0x5ddbb9b8200f]
[27] python3(+0x2805a5)[0x5ddbb9b875a5]
[28] python3(_PyRun_SimpleFileObject+0x1a8)[0x5ddbb9b86b88]
[29] python3(_PyRun_AnyFileObject+0x47)[0x5ddbb9b86867]
*** End of error message ***
Aborted (core dumped)
```

Implementation Overview
The core workflow follows:
```python
system = md.Create("./Mooring/lines.txt")

# Initialize nodes
vertices = np.zeros((num_nodes,3))
for i in range(num_nodes):
    point = md.GetPoint(system, i+1+num_nodes)  # Node indexing offset
    vertices[i] = md.GetPointPos(point)
pos_array = vertices.flatten().tolist()
vel_array = [0]*3*num_nodes
md.Init(system, pos_array, vel_array)  # Successful initialization

while step():  # Time-stepping loop
    md.SaveState(system, "oldstate")  # Save previous state
    
    # Exchange data with structural solver
    # ... (pos_array/vel_array updated from solver, current_time/dt defined)
    
    force_array = md.Step(system, pos_array, vel_array, current_time, dt)
    
    if not convergence():  # Rollback on divergence
        md.LoadState(system, "oldstate")  # Failure occurs here after ~2000 cycles

md.Close(system)
```
# Debugging Attempts
Verified array dimensions match MoorDyn expectations
Confirmed file system permissions for SaveState
Memory profiling shows no significant leaks before failure
# Environment
Python 3.9.18 
MoorDyn v2.3.8 (pip-installed binary)
OS: Ubuntu 22.04.3 LTS (32GB RAM)
Structural solver: Custom implicit FEM (details available upon request)

Any insights or debugging suggestions would be greatly appreciated. Thank you for maintaining this invaluable tool!
Best regards,
Wzlr

sanguinariojoe, 2025-03-22:
Saving and loading the same file too many times might not be a good idea (specially with python on some OSs). Please try replacing:

```
moordyn.SaveState(system, "oldstate")
moordyn.LoadState(system, "oldstate")
```

by

```
oldstate = moordyn.Serialize(system)
moordyn.Deserialize(system, oldstate)
```

On top of that, the error you have is related with the GIL (see https://github.com/python/cpython/blob/3.11/Python/ceval.c#L327). That problem might be solved in Python>=3.12 (if you look for `_Py_FatalError_TstateNULL` [here](https://github.com/python/cpython/blob/3.12/Python/ceval.c) you will find nothing)

wulizhiren, 2025-03-24:
@sanguinariojoe  Thank you for your insightful suggestions!

Following your advice, I:

Upgraded to Python 3.12 (previously 3.10)
Eliminated file-based SaveState/LoadState in favor of in-memory state management.

The combination resolved the crash at ~2000 steps perfectly. The performance gain from avoiding disk I/O is an unexpected bonus!

Your guidance not only fixed this issue but also improved our solver's efficiency. Truly appreciate your time and the community's support.

closed 2025-03-24 by wulizhiren