FloatingArrayDesign/MoorDyn #302: seafloor and wavekin are sometimes extremelly slow on GCC

sanguinariojoe, 2025-03-28:
When compiling with GCC (Release), the tests seafloor and wavekin might take several minutes. It is not happening always, but it is quite common. That is not happening with CLang.

I did not see any memory error, as far as I remember.

sanguinariojoe, 2025-03-28:
For instance, while installing with yay:

```
(...)
10/30 Test #10: tripod ...........................   Passed    5.06 sec
      Start 11: seafloor
11/30 Test #11: seafloor .........................   Passed  162.72 sec
      Start 12: wavekin
12/30 Test #12: wavekin ..........................   Passed  482.32 sec
      Start 13: wavekin_7
13/30 Test #13: wavekin_7 ........................   Passed    4.18 sec
(...)
```

RyanDavies19, 2025-03-31:
@sanguinariojoe, strange. I have not looked too much into it, but this summer NREL was hoping to work a bit on tests (mostly getting the MD-F/MD-C comparison to pass). We could try to tackle this if it is not figured out by then.

sanguinariojoe, 2025-03-31:
I am on my parental leave now, so I will not work on this.

I have been noticing this for a long time actually, so it is not because of
the recent changes