FloatingArrayDesign/MoorDyn #309: Question about 'Volume'

LJM-97, 2025-05-17:
Hi,

In `MoorDyn`, the `Volume` is interpreted as calculating buoyancy. When the initial moment corresponds to a stable operational state of the platform, this can be well estimated. However, how should `Volume` be defined if a free-decay test based on pitch motion is to be conducted?

[image]

Best regards!

sanguinariojoe, 2025-05-17:
Hey Jia-Ming!

I was planning to reply you on foamMooring's repo ASAP. But since you
raised the issue here, i can reply you by email now.

As Haifei told you, volume is ignored because the body is a coupled one,
i.e. is controlled by OpenFOAM. Moordyn is just feeding OpenFOAM with the
forces coming from the attached lines.

So you can just set 0.

P.S. The reason I am not so responsive is that I am on my parental leave.
Would you be so kind to reply yourself with this info on foamMooring's repo?

LJM-97, 2025-05-19:
> Hey Jia-Ming!
> 
> I was planning to reply you on foamMooring's repo ASAP. But since you
> raised the issue here, i can reply you by email now.
> 
> As Haifei told you, volume is ignored because the body is a coupled one,
> i.e. is controlled by OpenFOAM. Moordyn is just feeding OpenFOAM with the
> forces coming from the attached lines.
> 
> So you can just set 0.
> 
> P.S. The reason I am not so responsive is that I am on my parental leave.
> Would you be so kind to reply yourself with this info on foamMooring's repo?
> […](#)
> 
>  <[#309](https://github.com/FloatingArrayDesign/MoorDyn/issues/309)>
> 
>  Hi,
> 
>  In MoorDyn, the Volume is interpreted as calculating buoyancy. When the
>  initial moment corresponds to a stable operational state of the platform,
>  this can be well estimated. However, how should Volume be defined if a
>  free-decay test based on pitch motion is to be conducted?
> 
>  image.png (view on web)
> 
>  Best regards!
> 
>  <[#309](https://github.com/FloatingArrayDesign/MoorDyn/issues/309)>, or
>  unsubscribe

Thank you so much for your kind and helpful explanation! I'd be happy to post this clarification to the foamMooring repo on your behalf.

Furthermore, could the other entries in the `BODIES` section also be omitted?
What is the recommended ratio between `NumSegs` and the mooring line length `UnstrLen`?

Wishing you and your family all the best!

RyanDavies19, 2025-05-19:
Hi @LJM-97, 

I can jump in here for your questions. The other entries in the body section need to be there for the input file to be successfully read, but you can set them to 0 so they won't have any impact on the simulation. 

As for ratio of NumSegs to UnstrLen, it depends on your system. For longer lines (500-1000m UnstrLen), typically we use 25 - 50 m long segments. For shorter lines, we use closer to 5- 10 m segments. Generally I wouldn't recommend shorter than 1 m segments, as things can get unstable on that scale. 

A good rule of thumb for this is as follows: A number of segments is producing accurate results if you get the same response when running the same simulation with double the amount of segments. The less segments you have, the more efficient your simulation, so we generally try to use the smallest NumSegs that is accurate.