FloatingArrayDesign/MoorDyn #310: Significant Discrepancy in Fairlead Pretension Between MoorDyn_driver and Python Library

JT-Kim89, 2025-06-02:
Hello,

I’m currently working with the Python interface of MoorDyn, and I’ve encountered a significant discrepancy in the fairlead pretension results.

Using the OC4 Semi-submersible model, I tested the fairlead pretension with both:

- MoorDyn_driver ([reference tool](https://github.com/OpenFAST/r-test/blob/main/modules/moordyn/md_5MW_OC4Semi/driver.MD.out))
- A Python-based script (custom interface)

The result from MoorDyn_driver gives a pretension of approximately 1,100 kN, which is within the expected range. However, my Python script outputs a much higher value of 23,430 kN, which is nearly 20 times larger.

This leads me to suspect that there may be an error in my Python implementation. I’ve double-checked the input and configuration files and have linked them here for reference:

💬 If anyone has experience using the MoorDyn Python library and can help point out what might be going wrong, I’d greatly appreciate your support.

Thank you!

```
import moordyn

# Create MoorDyn system
filename = r'filepath\Moordyn_Driver\moordyn.dat'   # same input files of OC4 from moordyn driver
system = moordyn.Create(filename)

# Initialize position, velocity vectors
x =[0] * 9       # Position input [m]
xd = [0] * 9      # Velocity input [m/s]

moordyn.Init(system, x, xd)

n_lines = moordyn.GetNumberLines(system)
for line_id in range(1, n_lines + 1):
    print("Line {}".format(line_id))
    print("=======")
    line = moordyn.GetLine(system, line_id)
    n_segs = moordyn.GetLineN(line)
    ten = moordyn.GetLineFairTen(line)/1000
    print("  ten = {}".format(ten))

# Finalize MoorDyn system
moordyn.Close(system)
```

``` 
Line 1
=======
  ten = 23428.795419310234
Line 2
=======
  ten = 23428.609571003337
Line 3
=======
  ten = 23428.795419310234
```

sanguinariojoe, 2025-06-02:
Paste the input file, please.

Sound like an initialization issue. MoordynF (what you called driver) and
MoordynC have different initialization methods (with different strengths
and drawbacks).

See https://moordyn.readthedocs.io/en/latest/initialization.html

JT-Kim89, 2025-06-10:
@RyanDavies19  
Thank you for your help!
As you mensioned, after loading the fairlead positions instead of using zeros for initialization, the pretension results now closely match those from the MoorDyn driver. I really appreciate it! 👍👍👍 

```
import moordyn

# Create MoorDyn system
filename = r'filepath\Moordyn_Driver\moordyn.dat'
system = moordyn.Create(filename)

# Initialize position, velocity vectors
# Get the initial positions from the system itself
x = []
for i in range(3):
    # 4 = first fairlead id
    point = moordyn.GetPoint(system, i + 4)
    x = x + list(moordyn.GetPointPos(point))
print(x)

xd = [0] * 9      # Velocity input [m/s]

moordyn.Init(system, x, xd)
print(" - MoorDyn initialized.")

# Get number of lines dynamically
n_lines = moordyn.GetNumberLines(system)
for line_id in range(1, n_lines + 1):
    print("Line {}".format(line_id))
    print("=======")
    line = moordyn.GetLine(system, line_id)
    n_segs = moordyn.GetLineN(line)
    ten = moordyn.GetLineFairTen(line)/1000
    print("  ten = {}".format(ten))

# Finalize MoorDyn system & End time logging
moordyn.Close(system)
```

```
[20.434, 35.393, -14.0, -40.868, 0.0, -14.0, 20.434, -35.393, -14.0]
 - MoorDyn initialized.
Line 1
=======
  ten = 1117.2703627249816
Line 2
=======
  ten = 1117.1157089889437
Line 3
=======
  ten = 1117.2703627249816
```