FloatingArrayDesign/MoorDyn #313: Question About High-Frequency Noise in MoorDyn Tension Results (Python vs. Moordyn_Driver)

JT-Kim89, 2025-06-22:
Hello,

As before, I am building a MoorDyn model in Python using OC5 data. During the initialization phase, the pretension results matched quite well. However, I am currently performing a time-domain simulation based on prescribed 6-DOF platform motions to calculate mooring line tensions, and I am observing some discrepancies in the resulting tension responses. I would like to ask for your advice regarding these differences.

🔍 Test Setup

- Input: 6-DOF motion from OC5 regular wave case
- Comparison: Python MoorDyn vs. MoorDyn_Driver
- Observed Issue:
  - Overall tension trends are similar.
  - However, the **Python output shows significantly more high-frequency noise**.

🛠 What I Have Tried
- I applied the same filtering(forward_backward_filter) to the input motions and velocities as implemented in the Fortran driver.
- The input velocities were calculated using central differencing from the input motion time series.
- Despite this, the high-frequency noise in the Python tension results remains.
- I reviewed the Fortran source code but could not find any additional filtering applied to the tension output.

🙋‍♂️ My Question

- **Is it necessary to apply additional filtering to the tension output when using the Python version of MoorDyn?**

📎 Attachments

- Python script and input values used for comparison are included below (attached).
- Unfortunately, I am unable to include output plots due to company network restrictions.

Thank you as always for your kind support!
Looking forward to your advice.

```
while t <= time[-1]:
    # 1.  motions of the platform
    trans     = np.array([surge_pos[iStep], sway_pos[iStep], heave_pos[iStep]])
    trans_vel = np.array([surge_vel[iStep], sway_vel[iStep], heave_vel[iStep]])
    
    rot_angles = [yaw_pos[iStep], pitch_pos[iStep], roll_pos[iStep]]  # ZYX
    rot_matrix = R.from_euler('ZYX', rot_angles).as_matrix()
    ang_vel    = np.array([roll_vel[iStep], pitch_vel[iStep], yaw_vel[iStep]])

    p.clear()
    v.clear()
    
    # 2. positions, velocities of the fairleads
    for i in range(3):
        rel_pos     = initial_fairlead_positions[i] - platform_initial_pos
        rel_rot     = rot_matrix @ rel_pos
        global_pos  = rel_rot + trans
        global_vel  = np.cross(ang_vel, rel_rot) + trans_vel

        p += global_pos.tolist()
        v += global_vel.tolist()
        
    # 3. MoorDyn Step
    moordyn.Step(system, p, v, t, dt)
    
    # 4. Fairlead tensions
    tensions = []
    for line_id in range(1, moordyn.GetNumberLines(system) + 1):
        line = moordyn.GetLine(system, line_id)
        fairlead_ten = moordyn.GetLineFairTen(line) / 1000  # [kN]
        tensions.append(fairlead_ten)
```

```
--------------------- MoorDyn Input File ------------------------------------
Mooring system for OC5-DeepCwind Semi
FALSE    Echo      - echo the input file data (flag)
----------------------- LINE TYPES ------------------------------------------
Name     Diam      MassDen       EA    BA/-zeta    EI    Cd      Ca     CdAx   CaAx
(-)       (m)      (kg/m)        (N)    (N-s/-)    (-)   (-)     (-)    (-)    (-)
SBA       0.1369    125.6    7.520E8     -1.0        0          1.404    0.865    0.213     0.269
BOW       0.1398    125.8    7.461E8     -1.0        0          1.404    0.865    0.213     0.269
PSA       0.1393    125.4    7.478E8     -1.0        0          1.404    0.865    0.213     0.269
---------------------- POINTS --------------------------------                                           
ID     Attachment  X        Y         Z      M      V       CdA   CA
(-)    (-)        (m)      (m)       (m)    (kg)   (m^3)   (m^2)  (-)
1      Fixed    418.8    725.383  -200.0     0       0       0     0
2      Fixed   -837.6      0.0    -200.0     0       0       0     0
3      Fixed    418.8   -725.383  -200.0     0       0       0     0
4      Vessel    20.434   35.393   -14.0     0       0       0     0
5      Vessel   -40.868    0.0     -14.0     0       0       0     0
6      Vessel    20.434  -35.393   -14.0     0       0       0     0
---------------------- LINES --------------------------------------                                           
ID      LineType   AttachA   AttachB  UnstrLen  NumSegs   Outputs
(-)       (-)        (-)       (-)       (m)      (-)       (-)
1         SBA        1         4       835.5     20        -
2         BOW        2         5       835.5     20        -
3         PSA        3         6       835.5     20        -
---------------------- SOLVER OPTIONS ---------------------------------------
0.001   dtM       - time step to use in mooring integration (s)
3.0e6    kb      - bottom stiffness (Pa/m)
3.0e5    cb      - bottom damping (Pa-s/m)
200        WtrDpth      - water depth
RK4  tScheme      - Time integrator
2.0      dtIC      - time interval for analyzing convergence during IC gen (s)
60.0     TmaxIC    - max time for ic gen (s)
4.0      CdScaleIC - factor by which to scale drag coefficients during dynamic relaxation (-)
0.001    threshIC  - threshold for IC convergence (-)
------------------------- need this line --------------------------------------

```

sanguinariojoe, 2025-06-22:
It seems to me that you are evaluating positions and velocities
inconsistently.

When calling moordyn.Step() you shall pass the positions r(t_n), and the
velocities u(t_n+1/2). Notice that they are evaluated on different time
instants.

Along this line, you can fix the noise by simple consider the positions
collected from moordyn itself, using moordyn.GetPointPos() (check here:
https://github.com/FloatingArrayDesign/MoorDyn/blob/master/tests/minimal.cpp#L202).
Then you can probably improve the agreement considering as velocities
(r(t_n+1) - r(t_n)) / dt, with r(t_n) the position got with
moordyn.GetPointPos() and r(t_n+1) the position computed with your
algorithm, but considering iStep+1

Let us know please!

JT-Kim89, 2025-06-27:
Dear @sanguinariojoe,

Apologies for the late reply.

To get straight to the point — I was able to significantly reduce the noise **by using filtered input positions** instead of the raw motions. Previously, I had been feeding raw motion data directly into moordyn.Step(), which led to noticeable high-frequency noise in the tension results. (I wish I could share the plots, but unfortunately, due to company security policy, I’m unable to upload figures.)

Regarding velocity input, I updated it to approximate the mid-step velocity u(tₙ₊½) using the average of two adjacent steps as below:

```
# u(tₙ₊½) ≈ 0.5 * (uₙ + uₙ₊₁)
trans_vel = 0.5 * (np.array([surge_vel[iStep], sway_vel[iStep], heave_vel[iStep]]) +
                           np.array([surge_vel[iStep+1], sway_vel[iStep+1], heave_vel[iStep+1]]))
ang_vel = 0.5 * (np.array([roll_vel[iStep], pitch_vel[iStep], yaw_vel[iStep]]) +
                          np.array([roll_vel[iStep+1], pitch_vel[iStep+1], yaw_vel[iStep+1]]))
```
Although the impact of this update on the overall results was relatively small compared to position filtering, I applied it to ensure consistency with the expected formulation.

Thank you again for your kind guidance. 👍 👍 👍 
Have a great day! 

Best regards,
Jungtae Kim

closed 2025-06-27 by JT-Kim89