FloatingArrayDesign/MoorDyn #314: Validation data for seabed friction

kouki0000, 2025-06-24:
Are there any papers comparing the effects of seabed friction with model test results, etc.?

RyanDavies19, 2025-06-24:
Hi @kouki0000, 

There are a few papers in the references section of the docs that discuss seabed contact: https://moordyn.readthedocs.io/en/latest/references.html

The model implemented is described in this one: https://doi.org/10.1088/1742-6596/2362/1/012018

In short we verified the performance of our approach by comparing to an equivalent OrcaFlex simulation.

kouki0000, 2025-06-24:
Thanks for the reply.
I believe the OrcaFlex simulation has many users and is well tested.
However, I don't think the OrcaFlex simulation is well validated when limited to the effects of seabed friction.
Is there no model test results to serve as validation data, since the behavior of anchor chains in the water needs to be measured and model tests are difficult to perform?

RyanDavies19, 2025-06-25:
@kouki0000, 

I am not aware of any direct model tests that examined the approach in MoorDyn. The paper that describes the approach we use (https://doi.org/10.1088/1742-6596/2362/1/012018) has a pretty robust literature review of the different seabed friction modeling approaches and offers good comparison data. I will also note that the seabed friction modeling in MoorDyn is a very commonly used feature in numerous studies that look at mooring dynamics. We have (in work yet to be published) been able to effectively match the behavior seen in experimental results that included seabed friction, but that wasn't the main focus of the work.

closed 2026-02-17 by sanguinariojoe