FloatingArrayDesign/MoorDyn #315: Question about coupling current code in Rk4

frozenheartsaber, 2025-06-27:
Hello，

I am using Moordyn in matlab for rigid body modeling, and I am using a Runge Kutta 4th order time stepping method to calculate the euler equation.
I added the Moordyn in the beginning of each time step T currently as shown below:

for i = 1:totalstep
    T=double((i-1))*delta_t;%当前的时间
    [~,TopTension(:,i)] = MoorDynM_Step(system, x_moor(:,i), dx_moor(:,i),T,delta_t);
    xxxxxxxxxxxxxxxxxxx
end

As Runge Kutta contains k1,k2,k3,k4, I need to update the Mooring forece at T+delta_t/2 and T+delta_t to calculate the real solution, I hope to know if there are methods to update the Mooring force in K1 to K4 to make the coupling more accurate?

[image]

Thank you.

sanguinariojoe, 2025-06-27:
Just be careful, specially on the final update, or you'll end up with
inconsistent/incompatible motions

frozenheartsaber, 2025-06-28:
> Just be careful, specially on the final update, or you'll end up with
> inconsistent/incompatible motions
> […](#)

Hello

I tried to  use Serialize in following method:

1. At K1 which means t: 
[t0,TopTension(:,i)] = MoorDynM_Step(system, x_moor(:,i), dx_moor(:,i),time,0);
2. At K2 which means t+0.5*dt: 
    S0=MoorDynM_Serialize(system);
    [t0,TopTension(:,i)] = MoorDynM_Step(system, Moor_x_current, Moor_dx_current,time,0.5*delta_t);
    MoorDynM_Deserialize(system,S0);
3. At K3 which means t+0.5*dt again (as Rk4 needed): 
    S0=MoorDynM_Serialize(system);
    [t0,TopTension(:,i)] = MoorDynM_Step(system, Moor_x_current, Moor_dx_current,time,0.5*delta_t);
    MoorDynM_Deserialize(system,S0);
4. At K4 which means t+dt: 
    S0=MoorDynM_Serialize(system);
    [t0,TopTension(:,i)] = MoorDynM_Step(system, Moor_x_current, Moor_dx_current,time,delta_t);
    MoorDynM_Deserialize(system,S0);
I hope to get an output time series as: 0,0.05,0.05,0.1 and 0.1,0.15,0.15,0.2 ......
but the result I get from the output file shows :
0	 546519	 
0.005	 2.5944e+07	 
0.005	 2.59376e+07	 
0.01	 6.90953e+06	 
0.005	 2.59292e+07	 
0.005	 2.59227e+07	 
0.01	 6.9051e+06	

I hope to know how to fix that problem with serialize.

frozenheartsaber, 2025-06-28:
Hello

I tried to use Serialize in the following method:

At K1, which means time t:
[t0,TopTension(:,i)] = MoorDynM_Step(system, x_moor(:,i), dx_moor(:,i),time-delta_t,delta_t);

At K2 which means t+0.5dt:
S0=MoorDynM_Serialize(system);
[t0,TopTension(:,i)] = MoorDynM_Step(system, Moor_x_current, Moor_dx_current,time,0.5delta_t);
MoorDynM_Deserialize(system,S0);

At K3 which means t+0.5dt again (as Rk4 needed):
S0=MoorDynM_Serialize(system);
[t0,TopTension(:,i)] = MoorDynM_Step(system, Moor_x_current, Moor_dx_current,time,0.5delta_t);
MoorDynM_Deserialize(system,S0);

At K4 which means t+dt:
S0=MoorDynM_Serialize(system);
[t0,TopTension(:,i)] = MoorDynM_Step(system, Moor_x_current, Moor_dx_current,time,delta_t);
MoorDynM_Deserialize(system,S0);

I hope to know if this is the right way to use it. Another question is that, x_moor(:,i), dx_moor(:,i) are displacement and velocity as time t, but I have to use t-delta_t as the beginning time to get the results at time t. If there is a way to fix it?