FloatingArrayDesign/MoorDyn #317: questions about clump weight

Jinhezai123321, 2025-07-08:
Dear professor,I want to add a clump weight to the mooring system as shown in form2 but my counterweight doesn't seem to work, I don't know what went wrong, I guess it's probably the connection type.If not using "free" or "point," what keywords should be used instead?
If not using "free" or "point," what keywords should be used instead

[image]

[image]

mattEhall, 2025-07-08:
Hi, I'm not certain of the problem without knowing more about what didn't work, but I can offer a couple suggestions. Your current setup has 10 segments in the short lines between clump weights; that will result in extremely short segments of 1.5 m. Whereas line 1 has 18 m segments. Try for more consistent segment lengths in the mooring system.
You could use just 1-3 segments for the lines between clump weights. Or you could replace the whole clump-weighted portion of the mooring line with a heavier line object (that has equivalent weight-per-length as with thr clump weights).

Jinhezai123321, 2025-07-08:
> My suggestion is just about changing the NumSegs vales so that the segment lengths (i.e. UnstrLen/NumSegs) do not get very small. The connection type should be free, as you have it. The lengths of the lines (UnstrLen) also seem fine.

thank you very much，I will try this later

Jinhezai123321, 2025-07-08:
> My suggestion is just about changing the NumSegs vales so that the segment lengths (i.e. UnstrLen/NumSegs) do not get very small. The connection type should be free, as you have it. The lengths of the lines (UnstrLen) also seem fine.

Dear professor,In order to verify whether it is a segmentation problem, I first built a model without a counterweight, calculated the prestress and matched the results of the paper, but after I added the counterweight to the middle section, the prestress did not change, so there must be something wrong?

[image]

Jinhezai123321, 2025-07-10:
> Hi [@Jinhezai123321](https://github.com/Jinhezai123321),
> 
> It looks like the weight you added with point 2 is a `vessel` type, which is the same as `coupled` (vessel is a legacy naming convention we still support). If you want the point that attaches lines 1 and 2 to act like a clump weight then it needs to have the type set to `free`.

Dear professor,I was using the free type before, but when I use the free type, my mass doesn't seem to work When I calculate the pretension, I don't know what's wrong. Quality only works if I use the coupled type, but here I think free should be used

[image]

sanguinariojoe, 2025-07-10:
Do you get good pretension values if you set it as "fixed"?

Jinhezai123321, 2025-07-15:
Dear professor,i cannot get good pretension.Is there anything else I can improve on.

[image]

[image]

closed 2026-02-17 by sanguinariojoe