FloatingArrayDesign/MoorDyn #328: How does Step Calculation works

miriasmata, 2025-09-19:
Hi,

I am new using MoorDyn and I was following the example with the 3 mooring lines attached to a floating body. 

My main question is how the Step calculation works:

[image]

In that example, we estimate the system for 5 time steps by moving the body at 0.5 m/s in the positive x direction.

What is not clear to me is that for each time step, the values of x (position) and xd (velocity) are not updated. From my understanding, it always receives as input the vectors from the initial position.

I would like to understand this function as I would like to use MoorDyn in Python as a standalone version, and force the system by including waves and currents.

Thank you for the help

miriasmata, 2025-09-19:
Thank you for the fast response.

I understand now how the Step function works. 

Then my next question would be if MoorDyn also computes this resulting position. From the input file I believe I can get as output the position of each node of the line by adding a p:

[image]

Could I used this value as initial condition for the following time step?

My end goal would be to use the Water Kinematic functions available in MoorDyn-C. Can I force the model with WaveKin=3 or with Currents=2 and use the Step function with the position that MoorDyn computes?

I know MoorDyn can be coupled to other models, but I wanted to add the hydrodynamics using these available functions. 

Thanks,
Marisol

sanguinariojoe, 2025-09-19:
Short answer: No, MoorDyn is meant to be coupled with other solvers.

Long answer: Well, you can, but the model is definitely too simple for any
real practical application. Just replace the "coupled" points by "free"
ones (set them a mass, a volume, and drag and added mass coeffs). Then you
can call step() passing empty arrays to x and xd.

ZahidUllah33400, 2026-01-01:
@sanguinariojoe 
For better understand the MoorDyn, I have been running the 3-lines with Body Simple Example from matlab driver. I wanted to understand the fairlead forces and moments it provides. At first i ran the simulation using the input file given at MoorDyn_standalone_demo examples. I kept the body reference frame 70 below MSL. i got the fairlead forces and tension. Now for the 2nd time i defined the body reference frame to the MSL (0,0,0) and then defined the fairlead position in this body reference frame. This time i got different fairlead forces and tensions, which is a bit confusing for me. In both the cases, the applied displacement (x) and velocities vector (xd) were zeros. I was expecting same faireald forces but different moment (due to change in the body reference frame position). The simulation were run just for one time-step. Can you please point out any reason for different forces. Thank you.

sanguinariojoe, 2026-01-02:
> The input files and the matlab script are attached here. Appreciate it.
>
>    1.
>
>    Input file in which body is defined 70 below MSL.
>    Example_body.txt
>    2.
>
>    Input file in which body reference is defined at MSL.
>    Example_body_MSL.txt
>    3.
>
>    Matlab script
>    Matlab_script.txt
>    4.
>
>    output data comparison (fairlead forces and tensions)
>
> output_data_comparison.txt

Well, on the first example you shall pass x=(0,0,-70,0,0,0).

ZahidUllah33400, 2026-01-06:
@sanguinariojoe Thank you for your guidance. I tested the MoorDyn with the aforementioned input file and the results are perfect and stable. 
After fully understaning the MoorDyn, i try to use it for a different mooring system (which is my main target), however, i am facing some difficulties, and would appreciate you suggestion. I am using it for mooring system, consisting of 9 cables. I have attached the input file. i have two question if you can guide me.
i. I am running for t = 0, with no applied displacement (just want to see the tension in the initial configration of cables), however, i find that the tension changes with changing the dt. if dt  = 0.01, the tension is different, and if change dt, the fairlead tension changes. This was not the case in the OC3 system, where the tension remained same (for different dt) becuase no dispalcement or velocity was applied. 
ii. The tension seems lower than it is reported in the published article (i adopted the mooring system from a published article). Also the initilization report says, "Catenary initial profile failed". When i make the dt =0.1, then tension reache almost to the one reported in paper.

Any suggestion or possible reason, would be apprecaited. 

[Mooring_input_SJT_MSL.txt](https://github.com/user-attachments/files/24446408/Mooring_input_SJT_MSL.txt)

sanguinariojoe, 2026-01-07:
You can get your answers on the documentation:

 - https://moordyn.readthedocs.io/en/latest/troubleshooting.html#model-stability-and-segment-damping
 - https://moordyn.readthedocs.io/en/latest/initialization.html#an-initialization-practical-application